jakkra/Mars-Rover #1: Tubing

fabriziofiorucci, 2019-12-14:
Hello, thank you for sharing this very interesting project: far as I can see the rocker bogie tubing is not printed but looks like standard pvc tubing: what thickness did you select?
Thank you

fabriziofiorucci, 2019-12-14:
Solved! Measured from the f3d body

jakkra, 2019-12-14:
Hi thanks!

I was just about to send you this:
[image]
Yes it's some standard PVC pipe I found to save some printing time.
In Fusion 360 modify -> change parameters you can see the diameters.
I do hover doubt it's going to work just changing the setting there.

fabriziofiorucci, 2019-12-14:
Thank you! Just started printing the whole rover using asa-x filament

jakkra, 2019-12-14:
Awesome, good luck! I'm sure thats better than the cheap standard PLA I have printed with.

closed 2019-12-14 by jakkra

hoonkai, 2020-04-27:
@jakkra Hi What sort of adhesive did you use to attach the PVC and the parts together?

fabriziofiorucci, 2020-04-27:
@hoonkai in case this can help, please see here https://github.com/jakkra/Mars-Rover/issues/12#issuecomment-620130493

jakkra, 2020-04-27:
Hi @hoonkai, the fit is pretty tight but still needs something more to make it 100%, I just drilled a hole straight through and put a screw through like this: 
[image]