jakkra/Mars-Rover #2: Electronics schematics

fabriziofiorucci, 2019-12-15:
Hello, could you share the electronics schematics?
I’m building the electronics based on a raspberry zero and that would come handy.

Thanks!

jakkra, 2019-12-15:
I did draw a not very detailed, but good enough psuedo schematic a while ago. This was from before I added the arm and head. Now I have a I2C servo expander which all servos are connected to for saving pins on the ESP32. 
[image]

fabriziofiorucci, 2019-12-15:
That’s a good starting point: what ESCs do you use?

fabriziofiorucci, 2019-12-15:
I have a fully working design using 6 L298N drivers for the six dc motors, I’m evaluating whether to switch to ESCs

jakkra, 2019-12-15:
I'm using 2 of those with 3 motors connected to each of them
https://hobbyking.com/en_us/hobbyking-x-car-45a-brushed-car-esc.html
So I don't have control of each motor individually, only each side. 
I don't see a problem using 6 L298N as long as your DC motors doesn't draw more than 2A under load.

fabriziofiorucci, 2019-12-15:
Thanks: I actually am revamping a former rover I built using aluminium instead of plastic, and a lead acid battery instead of LiPO, with the aim of getting a self guided rover using a control software written in Java. Spare time permitting I might upload some contribution to add distance sensors, lidar etc.

jakkra, 2019-12-15:
Awesome! I'll also add more sensors after I finish the designing and printing the "back" parts of the rover.

I'm thinking of giving https://nasa.github.io/openmct/ a try for ground control software (the UI). I'm also looking for a new radio to get better range and to be able to stream low res video from ~1000m distance. Yet to find a suitable radio technology.

fabriziofiorucci, 2019-12-15:
The openmct project looks quite interesting: on my existing rover I added a GPS receiver and a digital compass and accelerometer, they work quite fine. By saving weight switching from aluminium to ASA and from lead battery to LiPO I might be able to put more stuff on it.
I'm experimenting a bit with LoRA for long range telemetry, but it's way too slow for video streaming, please keep me posted on your findings in that area.

fabriziofiorucci, 2020-01-13:
Hello, for the sake of curiosity, what kind of DC-DC stepdown converter do you use for servos? I'm running on MG996R that have a stall current of ~2.5A so I was wondering, additionally, if you're using one stepdown converter per servo or what else.

Thank you

jakkra, 2020-01-13:
Hi, I'm using variable buck converters like those: https://www.amazon.com/dp/B01MQGMOKI/ref=psdc_10967761_t1_B0758ZTS61
One for each servo (which is kind of a pain)

fabriziofiorucci, 2020-03-23:
[image]

Electronics work in progress...

- main board with Raspberry Zero, GPS receiver, magnetometer/accelerometer and ADC for distance sensors
- rocker bogie board with PWM controller for steering servos and control logic for L298N boards

jakkra, 2020-03-24:
Looking good! I finished re-routing my wires so now everything is inside. Currently in progress building a controller using Lora.

fabriziofiorucci, 2020-03-24:
LoRa is in my todo list: will you use it for both control and telemetry? Would you mind sharing some details? I guess you’ll use arduino/wemos, I’ll try the Raspberry way

jakkra, 2020-03-25:
My plan is to use it both for control and telemetry, however I have not yet experimented to see how much the lag is. The idea is to use WiFi control when the rover is close and  switch over to Lora when range increases. I'm going to use two TTGO LORA32, replacing the ESP32 dev kit in the Rover with this board with an ESP32 + Lora transceiver and the other board will be in the controller.

fabriziofiorucci, 2020-03-25:
That sounds really interesting. Please keep us posted!

fabriziofiorucci, 2020-03-29:
[image]
[image]

main board and rocker bogie board assembled and tested, ready to be installed

fabriziofiorucci, 2020-04-03:
[image]

Ready for testing

jakkra, 2020-04-07:
Looks clean! My current progress: 
[image]
Will have a tablet holder on top also for telematics display.

fabriziofiorucci, 2020-04-07:
it looks great! Will you share the f3d model?

jakkra, 2020-04-08:
Of courese!

fabriziofiorucci, 2020-04-08:
[image]

My high level electronics design, based on Raspberry Zero WH and I2C everywhere.

fabriziofiorucci, 2020-04-24:
[image]

Arm and head control board

fabriziofiorucci, 2020-04-24:
[image]

closed 2020-04-26 by fabriziofiorucci