jakkra/Mars-Rover #3: Steering ball bearing - best way to print

fabriziofiorucci, 2019-12-19:
Hello,

what's the best way to print the steering ball bearing? (see image) all in one print? body by body? I'm thinking to give it a try with 0.12 layer thickness and 30% infill. Did you print it or buy a ready-made one?

Thank you

[image]

jakkra, 2019-12-19:
Best way to print is like that. I printed with 0.2mm layer height. The infill I don't remember, but 30% sound ok. You might need to modify some clearences depending on your printer. I had to experiment a few times to get it as "stiff" but still rotating as possible. I think I have this bearing in a spearate Fusin360 project where you can modify some parameters. I'll do that when I get home tonight. But give it a try with current setup, it might work good for you.

fabriziofiorucci, 2019-12-19:
Thank you for the details, I’m printing on a creality ender 3, I’ll give it a try tonight.

fabriziofiorucci, 2019-12-20:
Hello again, would you have by chance che fusion project that allows changing clearances? I tried a few prints fiddling with slicer parameters with no success.
Thank you

jakkra, 2019-12-21:
Rename extention to .f3d from .txt as GitHub doesn't allow all file formats to be uploaded.
This is the bearing sketch, you can edit the parameters in the menu modify -> change parameters.
The base comes from thingiverse, but don't remember exact which thing. But here's the sketch.
[608zz10.txt](https://github.com/jakkra/Mars-Rover/files/3991388/608zz10.txt)

fabriziofiorucci, 2019-12-21:
Great, thank you! Meanwhile I’m at 25% of the overall rover body and rocker print. Hope to get the metal ball bearings soon. As soon as the electronics are fit I’ll post some more details.

jakkra, 2019-12-21:
Awesome, good luck!

closed 2020-01-07 by fabriziofiorucci