jakkra/Mars-Rover #4: Best support type to slice & print body

fabriziofiorucci, 2020-01-07:
Hello,

I'm proceeding with all prints, and when it comes to the body I wonder what's the most efficient support type to use when slicing stl files: I'm using Cura and I usually use the default line with 10% spacing. Any hint here/what have you used?

Thank you

jakkra, 2020-01-07:
I'm using Slic3r/PrusaSlicer, for supports I have just used the default ones in my slicer, so can't help you with which are best in Cura. Most parts don't need supports or just need some very simple ones. If you can give me a specific part, I can probably figure out how I printed it and send a image from my slicer.

fabriziofiorucci, 2020-01-07:
Hi, thank you. Most parts are perfectly designed to require a minimum amount of supports (good job!). The only parts I would need a hint is the main body (rover_body_front and the back part), it's quite large and I would like to minimize filament waste by optimizing the (large) support area.

jakkra, 2020-01-07:
I'll be home in an hour ish, will send you some images then.

jakkra, 2020-01-07:
No supports needed for the front, printed fine on my MK3, should be fine on your Ender 3 too.
[image]
Only required supports for back are those two.
[image]

fabriziofiorucci, 2020-01-07:
Thank you again! Cura tells me that I'm really hitting the width limit of the print bed and hence I should print it lying on one side. I'll try to fiddle with some parameters to print it the way you did. I'll keep you posted.

jakkra, 2020-01-07:
Yeah I printed it pretty much to be as big as I could print. You can in worst case just make the main rover body a cm och 2 shorter to make it fit.

fabriziofiorucci, 2020-01-09:
Hi, I managed to fix the printing area. I was wondering what infill and layer thickness you used. I would go for 20% and 0.24, that would be roughly 500g of filament and a couple of days of printing time.
Thank you

jakkra, 2020-01-09:
Great! I used 0.2mm layer thickness, and fairly low infill, maybe 5%. If i recall correctly the main body took about a day for each half to print.

fabriziofiorucci, 2020-01-09:
Thank you, I’ll try with that!

jakkra, 2020-01-10:
I looked a bit closer on my Rover and see that I used 3 perimeters for the bodies, instead of default 2.

fabriziofiorucci, 2020-01-10:
Thank you! I’ll change slicer settings accordingly

fabriziofiorucci, 2020-01-18:
[image]

currently printing the first half of the body: it takes the whole print area, thank you again!

fabriziofiorucci, 2020-02-22:
[image]

Body and rocker bogie prints came out perfect, next step: putting together the electronics

fabriziofiorucci, 2020-02-22:
Had some cracking issues while printing the body with ASA, so I switched back to PLA

jakkra, 2020-02-22:
Wow that's awesome! Feels weird to see it on other than my colors, but it looks really good 👍 
Good luck with the electronics, that's the easy part IMO :)

Here's a peek in what I'm currently working on, upgrading the wheels to flexible wheels for better grip (also upgradade to Rover 2020 look of the wheels, however I prefer the look of the old ones, but printing the old style in flex filament didn't turn out so good).
[image]

I also decided to finally clean up the wiring a bit, work in progress.
[image]

PS. I'll re-open this issue for better visablity for others :)

fabriziofiorucci, 2020-02-22:
Flexible wheels look great! I'm quite curious to see how ASA will react to friction when I start letting it run. Did you print the flexible part in TPU?

jakkra, 2020-02-22:
I think the ASA will do pretty well, I do not se much wear on my PLA wheels, and ASA should be comparable if not a bit stronger.
I actually don't know, bought it from a local store and it's unlabeled, but I would guess it's TPU.

closed 2020-04-26 by fabriziofiorucci