jakkra/Mars-Rover #5: Body drillings

fabriziofiorucci, 2020-01-14:
Hello,

the rover_back v1 f3d file shows a number of holes on the back part of the body (to attach reactor and antenna) and on the front part (arm). Since I'm quite new to fusion 360, how can the body STLs be exported, so that they include all holes?

Thank you

jakkra, 2020-01-14:
Hi thanks for finding this, looks like I forgot to do that as I just drilled those holes manually. I did not want to reprint the whole body. I have added the missing holes in this commit: 415d6b2ec4bc9af1005ce21a38afaa425a376768
Please check if I missed some.

I need to see if I can restructure the Fusion project a bit, I did not fully understand how to correctly work with components when I started which makes things like this those holes a bit wierd to add.

fabriziofiorucci, 2020-01-14:
Thank you, I'll check and will let you know

fabriziofiorucci, 2020-01-22:
Hi, the front part of the body with the arm hole printed perfectly in slightly more than two days. I'm now printing the rear part.
Thank you for helping!

closed 2020-01-22 by fabriziofiorucci