jakkra/Mars-Rover #6: what motors did you use

gitnem, 2020-01-22:
First off thanks for all the hard work you have put into this project.  I am interested in building this, but I am having issues sourcing a 25mm x 56mm brushless dc motor.  I would like to know which ones you choose to go with.

jakkra, 2020-01-22:
Hi, try searching for "brushless DC motor 12v 12v 60rpm" and you find some. Maybe this one:
https://a.aliexpress.com/_rvIAA0
Or this one: https://banggood.app.link/fPmdq9zPs3

However I see that the banggood one does not have a suitable rpm optional, but it might help you find one that will fit. The length of the motor is not that important, it will fit anyway.

You can also edit the Fusion design if you find some other motor you want to use.

fabriziofiorucci, 2020-01-22:
I got them from here:

€ 2,70  16％ Off | JGA25-370 Geared motor DC motor 6V 12V electric gear motor high torque 5/10/15/30/60/100/150/200/300/400/500/1000/1200 rpm
https://a.aliexpress.com/_UBuPo
 
You can select the rpm you need

gitnem, 2020-01-22:
thanks for all the info.  I noticed in another post you are using the https://hobbyking.com/en_us/hobbyking-x-car-45a-brushed-car-esc.html ESC, but your battery system is 3s (~12v).  The data sheets for that ESC say it needs 2s input, are you feeding 12v to the ESCs or do you have a step down?

Put another way, will feeding 3s to those ESCs harm them or not provide enough voltage to the motors?

jakkra, 2020-01-22:
Yes I know the ESCs are running out of spec, but they are very cheap and I have so far not had any problems using them at 12v. I'm sure it's not good to run them at 12V. The DC motors max out at less than 2 amps at stall, so that's max 6 amps and the ESCs are rated for 45A.

closed 2020-01-23 by gitnem