jakkra/Mars-Rover #8: Bearings

BarryReid, 2020-04-07:
I just bought some of the large size bearings and was wondering how people are securing them into position (main_leg_body_connector to the leg_boddy_bearling_connector).

jakkra, 2020-04-07:
For me the print is real tight so I had to really force the bearing in place so they sit there without anything additional. But if the fit is not perfect you could use some Loctite super glue, I use that in some places, for example to get the small bearings to stay.

fabriziofiorucci, 2020-04-08:
In case this might come handy: I printed the rocker bogie part using ASA-X and the body and arm with PLA. PLA prints (at least on my Creality Ender 3) came out perfect, whereas ASA-X shrunk a tiny bit while cooling down, even if my printer runs inside a temperature-controlled enclosure. In such a case the rocker bogie ball bearings were a bit loose. I solved that by simply designing and printing with PLA a very thin "adapter ring", basically a cilinder fitting the gap between the rocker bogie assembly and the ball bearing iself.

I preferred not to use glue in order to be able to disassemble parts in case I need to in the future.

I hope this helps.

fabriziofiorucci, 2020-04-10:
[Ball bearing adjustment.f3d.gz](https://github.com/jakkra/Mars-Rover/files/4460779/Ball.bearing.adjustment.f3d.gz)
[Ball bearing adjustment - steering wheels.f3d.gz](https://github.com/jakkra/Mars-Rover/files/4460780/Ball.bearing.adjustment.-.steering.wheels.f3d.gz)

In case this can help, here's the adjustment rings I designed.

Ball bearing adjustment.f3d can be used to fill the gap between the body and the main rocker bogie bearing
Ball bearing adjustment - steering wheels.f3d can be used to fill the gap between the four steering bearings and the rocker bogie printed part