jakkra/Mars-Rover #9: Grip rotator and servo insertion

fabriziofiorucci, 2020-04-09:
[image]

Hello,

I'm in the process of assembling the arm: when it comes to the grip rotator, how did you manage to install the servo? The print came out perfect but the servo doesn't slide in since the protruding rubber part where the power+pwm cable comes out leaves no space to insert the first part of the servo diagonally and then slide it in. Did you cut away some part somewhere?

Thank you

jakkra, 2020-04-09:
Ah yeah that one, I didn't think of that when designing it. What I ended up doing was to unscrew the "backplate" on the servo so that the wires wouldn't block and then put the servo together again after put in place.
By the way I didn't expect anyone to build the arm, so note that the design is far from perfect and has flaws. 

Also now while studying the cad I realised I missed fixing the design for a hole I missed, see attached image for where to drill manually. 
[image]
[image]

fabriziofiorucci, 2020-04-10:
Thank you! I'll assemble it later on today, the clamp works fine, just the right amount of friction. When I finish writing the software part I'm thinking about designing an interchangeable head in order to mount multiple tools like clamp, drill, laser head, etc.

fabriziofiorucci, 2020-04-10:
[image]

so far so good :-)

jakkra, 2020-04-10:
Awesome! I also shortened the PVC pipes quite a bit in the arm because the servos had some problems when they where as long as in the design. But since it's just PVC pipe you can easily experiment on what length works best for you, shorter => easier for servos. 

Interchangeable heads sounds awesome :) Really need to do the math for the arm (inverse kinematics) for it to work good, I haven't felt for doing it yet, but I hope I will some day.

fabriziofiorucci, 2020-04-10:
That's interesting: would you mind sharing the PVC pipes length you're currently using? So that I don't reinvent the wheel, I'm using al MG996R servos so they're pretty much standard.

jakkra, 2020-04-10:
Sure, both are 8 cm. I use the MG946R in the arm as they have about 13kg/cm more torque compared to MG996R with 11kg/cm, but have the exact same dimentions. I have also bough this servo http://www.goteckrc.com/ProductShow.asp?ID=52 for use here 
[image] which is 20kg/cm @7.4V, however I have not tested it yet. Should be alot stronger I think. That servo is the one with the most load.

Also note that there is difference between MG996R anf MG996R depending on brand, I have both from Towerpro and some clone brand, and the ones from Towerpro are way stronger, even if the models are the "same".

fabriziofiorucci, 2020-04-10:
Thank you, all of my servos are tower pro, I will see how they perform once assembled.

fabriziofiorucci, 2020-04-26:
[image]
[image]
[image]

Arm assembled, all parts came out perfectly.

jakkra, 2020-04-26:
Awesome! Looks cool! Now just some cable management, good luck :smiley: