jakkra/Mars-Rover #10: Servos don't fit

BarryReid, 2020-04-22:
Hi Guys

I bought TowerPro MG946R's for the steering as stated in the readme but after printing the wheel rotors they don't seem to fit at all. Just wondering if I am missing something.

jakkra, 2020-04-22:
The margins might be a bit tight, and differences between printers could cause servostyrning to not fit maybe. Is it way off, or could you sand the print a bit to get it to fit?

BarryReid, 2020-04-22:
Ye its quite a bit off and my printer is pretty accurate. I see in the cad model the server you have a model for is the MG996R not the MG946R like you have recommended in the docs. Could there be a difference there?

fabriziofiorucci, 2020-04-22:
Hello,

margins are a bit tight indeed, but the overall assembly has to be sturdy, so it makes sense. I use a Creality Ender 3 and in some cases my servos (TowerPro MG995R) fit perfectly, in some other cases I needed to use a dremel with a sandpaper tip to allow for a bit more space. I didn't print/use the wheel rotator, I decided to use the stock aluminium rotator that comes with the servo.

[image]
[image]

fabriziofiorucci, 2020-04-22:
[image]
[image]

BarryReid, 2020-04-22:
Ye those MG996R's must be a different size, you may want to update your doc on the front page as the MG946R's are quite a long way off (not just a snug fit) their mounting holes also don't line up.

[image]

jakkra, 2020-04-22:
You can just bend those "pins" off from the print. I did that now I remember. Will work anyway. Do they fit otherwise?

BarryReid, 2020-04-22:
No servers are still too big to fit even with a bit of force :) I will try shaving off some plastic from the part and see how its goes.

jakkra, 2020-04-23:
Like @fabriziofiorucci says, most parts in the rover work better if they are a tight fit. Some part might need some sanding to fit. If you are very far off, then come back and I'll update the model to give a litte more room for the servos.

fabriziofiorucci, 2020-04-23:
If I can add a bit/if this can be useful: I printed the rocker bogie assembly and wheels using ASA-X and my homebuilt heated printer enclosure (without it I can't simply print ASA-X due to cracking/warping) and after printing there is some shrinkage anyway, so steering servos were a bit tight (some sanding solved that).
When printing the body and arm I went for PLA and here warping/shrinking is virtually non-existing: when fitting the servos I needed to sand it even more if compared to the rocker bogie.

jakkra, 2020-04-23:
I will increase the margins around some servos next time I'm working on the Rover model (unless @BarryReid can't sand around it)

By the way a little teaser: 
[image]
[image]

fabriziofiorucci, 2020-04-23:
This is super cool! When are you going to post the f3d files? ;-)
At the end of the day did you go for LoRa+WiFi?

BarryReid, 2020-04-24:
Hey, ye if we could get a bit more space in the model that would be great, I need to sand so much that the walls of the print are almost non existent. I think this version of the servo is just a bit wider than the MG996R

BarryReid, 2020-04-24:
When I get home I will send you the measurements of the MG946R

jakkra, 2020-04-24:
Ok, that's a bit to much to sand, I will update the model tomorrow! I'll compare it with my servos and see how off it is.

For the controller, yes I went with both wifi and lora. Code, CAD and website for the phone is all about 80% done. Have not finished final design of the back of the controllers yet. I'll probably finish it up the next 2 weeks and make the repository public.

jakkra, 2020-04-25:
Realised the servo model I have used for reference is a about 1mm than the real one.
I measured my MG946R and MG99R and they have the exact same size. I made some more room in the "`original_wheel_rotor`" parts. Will those dimentions work for you? 
[image]
[image]

BarryReid, 2020-04-26:
Mate that's should be perfect! Did you make the change on the back and front rotors? I tried to download to check but cant see the updated CAD model in the GIT.

jakkra, 2020-04-26:
No, I didn't upload, wanted to get ok from you first. Will update asap.

BarryReid, 2020-04-26:
Thanks mate, did you make the change on the back and the front rotors?

jakkra, 2020-04-26:
Yes all 4.

closed 2020-04-26 by jakkra

BarryReid, 2020-04-26:
Hey @jakkra thanks so much for making those changes. I printed this part again overnight and have just completed a test fit. 

It now however seems to be a bit loose. Just wondering if you expanded it by a total of 1mm or if you did 1mm on each side?

I think if we made it half way between what it is now and what it was, it would be prefect.

Hope the feedback helps.

jakkra, 2020-04-27:
No, I did 0.5 on each side, so in total 1mm. If it can move inside then it's a problem, but otherwise I don't think your print is wasted, it should be hold in place when servo head I screwed in place. But anyway I'll update the model tonight with 0.25mm on each side instead of 0.5.

BarryReid, 2020-04-27:
Yip it moved ever so slightly. I think 0.25 on each side will be the perfect amount. The print should not be wasted as I will just glue in some single layer shims on each side of the servo. I will print the final rotor once you have updated the CAD and let you know the outcome.

Thanks

jakkra, 2020-04-27:
Feel free to print the final rotor now, I have now updated the dimentions.

k4m454k, 2020-05-23:
[image]

Same problem with TowerPro MG996R
[image]

BarryReid, 2020-05-23:
Ye, I just snapped them off, drilled some tiny pilot holes and screwed them in with a few self tappers.

k4m454k, 2020-05-23:
> Ye, I just snapped them off, drilled some tiny pilot holes and screwed them in with a few self tappers.

Barbarian =)

jakkra, 2020-05-23:
Yeah just snap them off, seems as the distance between the holes as a bit different between some brands of MG966R. They don't really do anything.

k4m454k, 2020-05-23:
> Yeah just snap them off, seems as the distance between the holes are a bit different between some brands of MG966R. They don't really do anything.  

I just redid this element and re-printed it.  the perfectionist inside me rejoices

[image]

jakkra, 2020-05-23:
@k4m454k hehe great! There are some things that are not perfect yet in the build, so you might need to compromise with your inner perfectionist some time later :wink: