jakkra/Mars-Rover #11: Software & communications flow

fabriziofiorucci, 2020-04-26:
I'm in the process of writing the control software.
This issue is mostly to share what I'm doing and to collect hints/other views on this topic.

Since my version of this rover is based on a Raspberry Zero WH, for development speed reasons I chose to use Java. I know that performance-wise and memory/CPU-wise it sucks, but there's a ton of ready-made libraries and I don't want to reinvent the wheel.
Namely, what I'm using is:

https://pi4j.com/ - for IMU, ADC, PWM controller, etc, it has lots of libraries ready to use
https://www.eclipse.org/paho/ - MQTT communication
https://eclipse-ee4j.github.io/jersey/ - REST API

I will eventually use WiFI and LoRa for the radio/control part, and all telemetry will be sent by the rover through MQTT. MQTT will also be used for interactive control.

Pre-programmed/batch control will be handled through a set of REST APIs that the code running on the rover will provide.

jakkra, 2020-04-27:
I think that's a great idea. When you have something working and if you release it in a public repo, I'll link to it in a new section in the README "Alternative SW" or similar.

fabriziofiorucci, 2020-04-28:
Hi @jakkra I just created a repo for my control software at https://github.com/fabriziofiorucci/Mars-Rover-Control-Software
As soon as I have an alpha release ready I'll update the repo.

jakkra, 2020-04-29:
Cool!

fabriziofiorucci, 2020-04-30:
Here you go: the pre-alpha release of my repo is ready, still work in progress

https://github.com/fabriziofiorucci/Mars-Rover-Control-Software

jakkra, 2020-05-17:
Nice progress on your control SW!
I just finished my Controller if you are interested https://github.com/jakkra/RoverController

fabriziofiorucci, 2020-05-17:
That looks really interesting! I just completed the first round of driving tests including interactive control (through SSH to the onboard Raspberry PI for now) and through REST API, driven by Postman.

I'm cloning your controller repo and I'll definitely try to build it. Keep up the great work!

Smarty-pro, 2020-07-13:
I don't understand, the 6 Channel RC receiver and transmitter is not enough to control the rover?

jakkra, 2020-07-13:
@fabriziofiorucci Is writing his own controller software for the rover. A 6 ch RC receiver/transmitter is enough for basic control of the rover. Without a microcontroller you are limited to how advanced your RC controller is, i.e. turning on the spot may not be possible.

Smarty-pro, 2020-07-13:
Oh nice

fabriziofiorucci, 2020-07-15:
FYI I just completed printing and adding the head, I'm working on the next commit of my control software + testing LoRA boards to control it remotely.