jakkra/Mars-Rover #12: Tubes

k4m454k, 2020-04-27:
What tubes are used for suspension?  I measured the diameters, 25x22.6 (external / internal).  but I did not find such in stores.

jakkra, 2020-04-27:
Hi, I posted the exact https://github.com/jakkra/Mars-Rover/issues/1#issuecomment-565747755 here: https://github.com/jakkra/Mars-Rover/issues/1#issuecomment-565747755 I bought the PVC pipe in a local store, so I can't really point you anywhere unfortunately. If you find similar pipe, you can edit the dimensions in the model yourself.

fabriziofiorucci, 2020-04-27:
Hi @k4m454k I used regular pvc tubes with an internal diameter of 20mm, I worked a bit with a dremel and sandpaper head to enlarge the tube's inner diameter and to scratch a bit the outer diameter of the printed parts. It took a while but I didn't want to print everything again and, on the nice side, the fittings are extremely tight, and the overall rocker bogie is very sturdy.
I hope this helps.

k4m454k, 2020-04-28:
Great! Thank you!

closed 2020-04-28 by k4m454k