jakkra/Mars-Rover #17: How to print head top connector & head rotator

fabriziofiorucci, 2020-06-22:
I'm trying to figure out how to print the head top connector & head rotator: I see that some bodies need to be exported as a single stl. One thing that is not clear to me is what bodies need to be exported/printed together in the red area here below:

[image]

Any hint? Thank you :)

jakkra, 2020-06-23:
Just print the parts seperate and glue them together. I also removed some parts that I did not end up using, so please refer to updated CAD model in 1446ce1866cb28418cfaf15fcf79e33b03aeff84
[image]

fabriziofiorucci, 2020-06-24:
That's clear now, thank you!

closed 2020-06-24 by fabriziofiorucci