jakkra/Mars-Rover #21: Printing tolerances

androiddrew, 2020-08-03:
Hey @jakkra since you provide the CAD model only I was wondering. Did your design have enough tolerance in the CAD model for direct printing or would I need to account for shrinkage in my slicer?

jakkra, 2020-08-03:
You should not need to shrink parts in your slicer, some parts are a tight fit by design. You may need to sand some parts a bit for perfect fit. Of course it depends a bit on printer calibration also, I print on a Prusa mk3.

So my recommendation is to print as is and then sand if needed, should not be too much.

androiddrew, 2020-08-03:
Ok, sounds great. I was referring to the shrinkage of the plastic as it cools. So printing at 101% scale, which roughly accounts for PLA shrinkage.

jakkra, 2020-08-03:
I think you should print at 100%, I have printed all parts in PLA also so it's designed for that.

closed 2020-08-05 by androiddrew