jakkra/Mars-Rover #23: Can the design drawing of the car provide a SolidWorks version

bluesky-ryan, 2020-08-30:
Dear jakkra:

   Thank you for your selfless sharing. Your rover is very cool. I also want to make one, but Fusion360 is very slow in China, and the license is basically not available. Is there a solidworks version of the car? Whether the bearings on the 4 driving wheels are missing, I wondered how to install them for a long time.

jakkra, 2020-09-02:
Hi, unfortunately from what I know that's not possible. What is possible is to export the Rover in .step format, I think Solidworks should be able to handle that. 
From my quick google search Fusion 360 should be fine to use in china and the free hobby license also valid in china. However I tried to export the whole rover as .step so you can try to import that, that's the best I can do. [rover_full.zip](https://github.com/jakkra/Mars-Rover/files/5164127/rover_full.zip)

jakkra, 2020-09-02:
For the bearing see this screenshot:
[image]

bluesky-ryan, 2020-09-03:
thank you very much!!! love you

bluesky-ryan, 2020-09-08:
Dear jakkra:

    What are the dimensions of these joints? I can’t draw these joints without dimensions.

[image]

bluesky-ryan, 2020-09-08:
This one I painted is very ugly, and the size is not right at all

[image]

jakkra, 2020-09-08:
I thought dimentions was included in the .step files. Are you sure you cannot use some measure tool in Soldiworks?

bluesky-ryan, 2020-09-09:
I tried the measuring tool inside, but this kind of irregular curved surface cannot be measured.，very sad. Is it necessary to use the space triangle Pythagorean theorem to calculate the size, mainly to keep the 3 wheels in a straight line and the middle wheel is exactly in the middle?

jakkra, 2020-09-09:
The middle wheel is not in the exact middle. Are you re-drawing the rover in Solidworks? I don't really follow. I can take some measurements if you specify exactly which distances you need.

bluesky-ryan, 2020-09-09:
Yes, on the one hand, some parts have different sizes and need to be redrawn. On the other hand, I also hope to be able to draw pictures by myself like you. I am learning to draw. You are now my idol. The design of the wheel support frame of your car is very ingenious. I have watched it for 2 days and cannot understand your design idea. How do you ensure that the wheels are in the same straight line? The support rods and support points of the rear wheels are not in the same line. This design in a straight line feels very cool, but I can’t understand how you can ensure the exact dimensions.

jakkra, 2020-09-09:
It's not that hi-tech, I just modeled from the ground up:
Created 1 wheel and wheel base, copied it 2 times and put them in a straight line, now we have 3 wheels in a straight line.
Then model the pipe in a direction so it meets the other pipes. Put a large square that connects the pipes coming from each wheel and TADA.

bluesky-ryan, 2020-09-10:
lol..   I thought you designed this frame according to a certain kinematics principle, doing complex calculations, and then started drawing parts according to this frame

Smarty-pro, 2020-09-22:
@bluesky-ryan this is what NASA did, this rover is an imitation of the famous Curiosity rover