jakkra/Mars-Rover #25: software

Albertogalvani, 2021-04-25:
Hi, i really like the rover, Nice project! 
 is there a guide for people that does not know how to program (like me) to control the rover remotely?

jakkra, 2021-04-25:
Hi, there is no guide, but it not hard to wire it up to a standard RC receiver. That is how I initially ran it (without the the arm). For steering and driving a 6 channel receiver is enough, 4 channels for the four steering wheels, and one channel each for two DC motor ESC:s (one for each 3 side motors). There are a lot of guides on Google and YouTube for setting up a RC transmitter/receiver.

Ivane55, 2024-09-22:
Hi, you have a very interesting project, I want to repeat it, but I'm not strong in programming. As I understand it, in your src folder there are only library files, is there an "ino" file somewhere for flashing the esp 32 module or is the module flashed in some other way?

jakkra, 2024-09-22:
Hi @Ivane55, this is an old project of mine and I don't really remember. Not that long ago another guy built a rover and he documented it a lot better, so I suggest go looking there instead: https://howtomechatronics.com/projects/diy-mars-perseverance-rover-replica-with-arduino/

Good luck!

closed 2024-09-22 by jakkra